gnubee-git/GnuBee_Docs #68: Debian modules

leocrawford, 2018-02-21:
How are the debian modules built? I can't find this documented? And in particular how to add extra kernel modules (I've had various problems getting ipsec and xtables working)

noblepepper, 2018-03-02:
I imagine Douglas cross compiles them, shouldn't be too hard once you get the cross compiling tools setup which may be tricky. I just compile stuff on the gnubee itself, to me it is easier albeit slower.

I haven't compiled a kernel but if you follow [this howto](https://github.com/gnubee-git/GnuBee_Docs/wiki/BuildOpenMediaVault) up to the first command in step 5 and add [module-assistant](https://wiki.debian.org/ModuleAssistant) to the packages installed the standard procedure using module assistant _should_ work

noblepepper, 2018-03-03:
I gave it a whirl, modules-assistant needs a non existent kernel headers package. I'll figure out how to get headers in place when I find some more time.

leocrawford, 2018-03-04:
Thanks for the tips. Had previously got stuck with kernel headers using a different approach. Even building kernel from source code have had to copy kernel headers manually