gnubee-git/GnuBee_Docs #70: Got the project abandoned?

Collax, 2018-03-05:
- `sh debian-jessie-install` does not work when executed with sh:
```
Choose partition to install (will be formatted !!!): /dev/sda2

/dev/sda2: LABEL="GNUBEE-ROOT" UUID="0b662189-70ac-4af1-8e79-cf00a6c45f61" TYPE="ext4" PARTUUID="7b866b93-dfb3-477c-bce5-4a2deaf03c7a"
Partition will be formatted, please type CONFIRM if you agree: CONFIRM
debian-jessie-install: 53: debian-jessie-install: [[: not found
Operation cancelled.
```
- The link in **debian-modules-install** pointing to the modules is dead (404) and so are the links in the documentation: https://gnubee.org/install_debian.html

- Overall the documentation is in rather poor condition.

Is on of the founders still active, working on the project?

thinkmassive, 2018-03-15:
The path was updated to reflect multiple GB-PC models. The new path is:
https://raw.githubusercontent.com/gnubee-git/GnuBee_Docs/master/GB-PCx/scripts/jessie_3.10.14/debian-jessie-install

(note the change from GB-PC1 to GB-PCx)

neilbrown, 2018-03-19:
Use "bash debian-jessie-install".  You probably has /bin/sh linked to dash.

Qwertie-, 2019-03-05:
I had issues getting the debian installer script working as well. Eventually I worked out how to get a working system are recorded the steps here https://itscode.red/posts/gnubee-install-guide/ 

Hopefully that should be some help.

neilbrown, 2019-04-01:
Thanks for that writeup.  Unfortunately it is already a little bit out-of-date, but that is so often the case on the Internet!!  Maybe we should put some of that detail on the wiki where it can be kept up-to-date.

Qwertie-, 2019-04-01:
A wiki page would be very useful. The install process took me ages because of a lack of information.

thinkmassive, 2019-06-05:
I received this email from the creator on May 14. It does not say the project is being abandoned, but he is trying to get rid of remaining stock. His contact info is at gnubee.org
——

Thank You for making the GnuBee Personal Cloud project a success!

Now offering reduced pricing on the remaining Personal Cloud Stock.

Personal Cloud One (PC1) 119 USD 
Personal Cloud Two (PC2) 149 USD
Personal Cloud Units come with side plates, brackets and screws.
Other items are available separately with an order :

PC1 Power Supply 10 USD [very limited supply]
PC2 Power Supply 18 USD
USB to UART Adapter 5 USD [limited supply]

Items are located in the U.S.A.
Shipping is not included at these prices.