gnubee-git/GnuBee_Docs #71: How to coerce wget in checking certificates?

Collax, 2018-03-05:
I have the packages **ca-certificates** and **openssl** installed and still wget is not using them to check the validity of sites delivered over https. How do I make it doing so?

phipolis, 2018-03-15:
Check out [this page](https://wiki.openwrt.org/doc/howto/wget-ssl-certs#certificate_directory) from the OpenWRT wiki.

phipolis, 2018-03-15:
You didn't say, but you also need to install wget, the stock version usually doesn't support ssl at all.

neheb, 2018-03-16:
If LEDE or recent OpenWrt trunk is being used, wget is actually uclient-fetch. By default it does not support HTTPS. uclient-fetch by default verifies certificates. --no-check-certificate must be passed to not do so.

I have no idea about stock or busybox wget.