gnubee-git/GnuBee_Docs #73: Can't shutdown without rebooting

phipolis, 2018-03-15:
I'm interested in using a UPS with software like apcupsd on a Debian GnuBee to detect a power failure and shut down cleanly. Unfortunately, it seems when the system goes down from a command like `shutdown -h now` or `halt`, the firmware starts it back up. 

Is there any way to get a GnuBee to stay down after shutdown until the power cycles?

neheb, 2018-03-16:
Write a script that kills most processes and unmounts most or all filesystems I guess.

To answer your question, no.

neilbrown, 2018-03-19:
> To answer your question, no.

True, but that's a bug. ralink_halt() in arch/mips/ralink/reset.c should have something link "while(1);" in it link ath79_halt() or ar7_machine_halt() or many others.

neilbrown, 2018-05-13:
This bug is fixed in mainline, and in my new 4.4 kernel firmware described at https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw
If you would particularly like it fixed in some other firmware, please specify which - no promises though.

phipolis, 2018-05-20:
I finally got a chance to test it and it works perfectly! Thank you so much for all your efforts!!

neilbrown, 2018-05-20:
Awesome.  I'll close this issue then.

closed 2018-05-20 by neilbrown

ulaker, 2018-06-10:
wake on lan work after successful shutdown working then?

neilbrown, 2018-06-10:
> wake on lan work after successful shutdown working then?

Very unlikely.
wake-on-lan is for waking up the system when it is in a low-power
state.  "halt" is for telling the system to stop doing anything
at all.  That might enter a low-power state, or might not.

I have seen no evidence that the gnubee hardware supports wake-on-lan.
As it is generally low-power hardware, there probably isn't much saving
in trying to go to sleep at all.

NeilBrown