gnubee-git/GnuBee_Docs #75: Status of mainline kernel support

sumpfralle, 2018-03-18:
I guess, it is of vital importance for the longevity of this project (and the hardware) to bring support the peripherals of GnuBee devices to the upstream kernel development tree. Otherwise we will end up with an outdated kernel (and thus more electronic landfill) somehwen ...

Since this is a not a short-term task, I propose to use this ticket for collecting the status of the ongoing mainlining efforts.

Could someone please start with summarizing the current state?
Thank you!

neilbrown, 2018-03-19:
https://github.com/neilbrown/linux/commits/gnubee/v4.15  mostly works.  I just today noticed that only the first SATA controller works.  Might fix that tomorrow.
I've posted some patches for inclusion in drivers/staging: https://lkml.org/lkml/2018/3/14/1035  Will repost with some improvements tomorrow.

sumpfralle, 2018-03-28:
Thank you for your quick response, for all the work you have put into mainline support and for the entertaining [lwn article](https://lwn.net/Articles/743609/) you wrote.
Your progress is way better, than I expected - this revives my positive emotions for the GnuBee platform. Thank you!

Adirelle, 2018-03-29:
@neilbrown Have you hit [this bug](https://github.com/gnubee-git/GnuBee_Docs/issues/54) with the ethernet driver ?

Right now, I have a job that checks the network connectivity every minute and reboot the GB if it fails...

alethiophile, 2018-03-30:
My experience is also that when it's booted into the Linux kernel, the network crash problem goes away. I see the problem when the board is in any of the three states uboot prompt/initramfs prompt/halted, but powered on.

Adirelle, 2018-03-30:
Weird. It always happens on booted kernel (the librecmc-based one, provided by GB) and does not take the switch down. The GB still have ICMP (e.g. ping works) but the TCP and UDP become unusable. That how  the jobs detects a failure : if it can ping a fixed IP but cannot connect to its echo service, it causes the system to reboot.

neilbrown, 2018-04-01:
Maybe there are two different bugs here - one that kills the switch and one that kills TCP but leaves ICMP working....

xvybihal, 2018-04-12:
@Adirelle I also hit bug #54 with this mainline kernel. GnuBee PC1 is sitting here on my desk, working "fine" (quotes because of ping shown later). But when I come back some hours/day(s) later, I can not connect to it - the time when it becomes unavailable via network is random, as far as I can tell. I Have to poweroff and power on again, nothing else worked for me ((un)plug cable).

```
gnubee ~ # uname -a
Linux gnubee.jvi.cz 4.15.12+ #3 SMP Wed Apr 4 15:29:16 CEST 2018 mips GNU/Linux

gnubee ~ # cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

gnubee ~ # cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet static
	address 172.16.202.254
	netmask 255.255.0.0
	gateway 172.16.whatever

```

Strange thing is, that when I run ping long enough, I see that every other while it takes gnubee quite long time to respond (100+ ms). I can **deffinitely** rule out faulty switch or cable.

https://gist.githubusercontent.com/xvybihal/b403c3c2677e423f3aea73feb8255a91/raw/d137ab85de894e2d5e3cb5bb9e2ffae16d023231/Gnubee%2520ping

Unfortunately I do not have UART cable to check what is going on, when GnuBee is not available via network.

This "thing" is nothing new, it was acting this way from the begining, with every kernel provided, and Deabian installed. Makes GnuBee pretty unusable for me, if I have to restart it several times a day.

neilbrown, 2018-04-12:
Thanks for the detailed bug report.  I love details.
What is happening on the gnubee before it starts failing?
Is it just sitting there idle?  Is it initiating network traffic itself?  It is responding to requests?
Are there lots of requests, or occasional? or few?  What protocol or protocols?
I've started a job fetching a 10K file over http every 2 seconds.  I'll see if it is still working in the morning.

Adirelle, 2018-04-12:
It does not seem to be linked to network activity. Stopping the transmission daemon to reduce the bandwidth and number of connections does not prevent it to happen (neither postponing it).  And I have downloaded an Ubuntu DVD torrent over a 1Gb link without any issue.

neilbrown, 2018-04-12:
So what *is* your gnubee doing when the problem happens?  How do you first notice it?  How long after boot does it typically fail?  <1hr?  <12hrs?
If I can reproduce the problem it is very likely that I can fix it
If not ... there are some kernel messages in that other bug which might be enough of a hint.  Maybe I'll try copying a bigger file.

xvybihal, 2018-04-13:
In my case, the GnuBee is doing nothing when it happens. What do I mean by nothing? The system installed is clean, no additional network daemons or services are installed. No network transfers are happening (except of time sync, and common stuff in the base system). I do not use it, it just sits here and doing nothing. The time when it happens is just random. Sometimes its hours, sometime day or two. 
Next time it happens, I will try to take out the SD card and copy some log here, thats probably the only thing I can do, without UART cable (which I am going to enquip myself with soon).

xvybihal, 2018-04-13:
Looking at the log/messages, there might be something helpful. Uploading whole file - it looks pretty similar with what @Adirelle posted [messages.tar.gz](https://github.com/gnubee-git/GnuBee_Docs/files/1906029/messages.tar.gz)

Adirelle, 2018-04-13:
> So what is your gnubee doing when the problem happens? How do you first notice it? How long after boot does it typically fail? <1hr? <12hrs?

Well, it seems totally random. The gnubee could be idle or I could be using it. Right now, it is running the following services : kernel nfs server, dropbear, dms (a upnp media server), ntpd, mysqld, transmission-daemon, ypbind, and there is a job running rdiff-backup once a day.

Sometimes it can be ok a whole week, sometimes only a few hours. I have the impression that the uptime rarely gets over 48 hours.

> If I can reproduce the problem it is very likely that I can fix it

I understand that -- I am a developer myself -- but unfortunately I could not find something to trigger the bug. I have only got the kernel error message.

I have a script that reboots it within one minute after it happens. It could dump the result of some commands in a file before rebooting, if you had some suggestions.

neilbrown, 2018-04-13:
Thanks for posting the whole log file (did I say that I love details?)
The problem appears to be that interrupts get disabled on one of the CPUs - presumably CPU0 as that gets all device interrupts on my gnubee..  This is reported by RCU at timestamp 2179 which should be about 21 seconds after it happened.  Then at timestamp 6317 (about 1 hour later) the network watchdog complains.
With interrupts disabled on one CPU, the network card will become unreliable.  As polling is sometimes used you can still get some traffic through, but it won't be fast.  A 'ping' probably works because it sends a packet every send, and only wants one in reply.  If it checks for incoming packets whenever it transmits a packet (quite likely), it will appear to work normally.
The next question is: why do interrupts get disabled? Two things might be useful.
1/ @Adirelle if you could get your script to "echo t > /proc/sysrq-trigger" when the problem is detected, that might help.  It should write stack traces for all process to the kernel log.  Seeing those would be most helpful.
2/ @xvybihal If you could rebuild your kernel with LOCK_DEP enabled, that might produce more useful info (I hope).  You need to enable CONFIG_PROVE_LOCKING and CONFIG_DEBUG_LOCKDEP and you may as well add CONFIG_PROVE_RCU.  Then if it happens again, collect the logs the same way that you did before.
Thanks.

neilbrown, 2018-04-14:
I have found a smoking gun.... I don't know if it is *the* smoking gun.
I went looking for places in the code which disable interrupts without clearly re-enabling them.  I didn't expect to find any - that would be too easy.  I found two!
One is almost certainly of no consequence - it always happens very early during initialization, so somethings must re-enable them.
The other is in the network code.  I don't understand the code enough to know when it happens, but it looks like it is in response to some event like the cable being pulled (though I tried that and it doesn't trigger anything).  The same bug is in the 4.4 kernel code
I've push out a fix to my gnubee/v4.15 kernel branch.  Please test in you can.

neheb, 2018-04-14:
@neilbrown Have you tried patching the stock Mediatek MMC driver to add support for mt7621? I tried and failed. Something about missing pinctl. I think I needed to edit the dts file...

This was my basis: https://github.com/jonpry/openwrt_mt7688/commit/a85e6d99899f3dc1204cd5bfba944e17bfa6178f

neilbrown, 2018-04-14:
> @neilbrown Have you tried patching the stock Mediatek MMC driver to add support for mt7621?

No I haven't.  As MMC is currently working, that driver is not a priority for me.  My current priorities are roughly:
- reboot - kernel currently leaves NOR flash in an inconsistent state
- SATA access seems really slow - 10MB/sec!
- network switch - enable VLAN support so different ports can be on different subnets
- 2nd network interface: the SOC has 2 interfaces to the switch, only one works at present
- crypto engine
- PCI driver has horrible hacks to select correct interrupt

So I won't be focusing on MMC for quite a while.  I'd be very happy for you or anyone else to dig into it and ask questions.  If you have specific focused questions, I'd be happy to share any expertise I might have.  I'd suggest opening a separate issue for each driver.

Adirelle, 2018-04-15:
@neilbrown

Compiled your v4.15 and it boots (which is already something since I was not sure about what I was doing) but :
* the initramfs does not find my root partition, since it lies on a md RAID1 array.
* it messes with the attached switch, probably because its own switch is not configured.

Your kernel branch might or might not fix some bugs, but as long as it works, I prefer having a recent version quite easy to build over the ones from libreCMC/LEDE. I have tried

Adirelle, 2018-04-15:
Got it working ; I was also missing some other modules. I will let you know if the network lockup happens again.

neilbrown, 2018-04-15:
Another convert - hurray :-)
I've merged your patch - thanks.

Adirelle, 2018-04-16:
Ok. So on the bad news:
* clock skew is back again (like https://github.com/gnubee-git/GnuBee_Docs/issues/49#issuecomment-347436829).
* performance seems lower than with the provided kernel, but I am not sure why.

I will try to compare kernel settings with the ones from LibreCMC.

Adirelle, 2018-04-16:
@dgazineu Is there a way to flash the firmware from a running Linux ? (to shorten the whole "put your image on a USB stick, plug it on the GB, reboot, let uboot flash the firmware, remove the USB stick, reboot" cycle).

neilbrown, 2018-04-16:
> clock skew is back again (like #49 (comment)).

You need the cpuclock clock-frequency in arch/mips/boot/dts/ralink/gbpc1.dts to match the value set by the u-boot that you have installed.  I have a u-boot that configures 900MHz so I set clock-frequency to 900000000.

> performance seems lower than with the provided kernel, but I am not sure why.

I just discovered that large kernel modules hurt performance.  I rebuilt with CONFIG_XFS=y (instead of =m, in O/.config) and filesystem throughput is a lot faster.  I've pushed an update for the defconfig file.

>  Is there a way to flash the firmware from a running Linux ?

Probably, using flash_erase and flash_cp from mtd-utils.  I haven't played much with them - be careful.
I set up my linux desktop as a tftp server and test kernels using tftpboot.  It is fairly painless.

Adirelle, 2018-04-16:
> I have a u-boot that configures 900MHz so I set clock-frequency to 900000000.

Ah, I think mine is configured at 880Mhz. It seems there were some discrepancy between shipped u-boot and the kernel updates that were provided later.

> I set up my linux desktop as a tftp server and test kernels using tftpboot. It is fairly painless.

To be sure I understand you right : once the new image is available through TFTP, you use the provided u-boot menu to download it and flash it. Or do you run the kernel without flashing it (which would be ideal) ?

neilbrown, 2018-04-16:
flash_erase works as expected.  flash_cp doesn't.  Something wrong with the spi driver...

neilbrown, 2018-04-16:
I don't need to flash the kernel to test it.
The script I use to build the kernel finishes with

> cp O/arch/mips/boot/uImage.bin /srv/tftpboot/GB-PCx_uboot.bin
> echo 'tftpboot;bootm 80200000'

I reboot (or power-cycle) the gnubee and press '4' repeatedly during the early messages.  Once is probably enough, but more doesn't hurt.
When I get the prompt I cut/paste that last message printed by the script.
For this to work you at least need to set serverip in your u-boot environment
e.g.

> setenv serverip 192.168.1.4
> saveenv

You might also want to set, or at least check (printenv) bootfile, and ipaddr

Adirelle, 2018-04-16:
> echo 'tftpboot;bootm 80200000'

Er... from where have you got the 80200000 ?

neheb, 2018-04-17:
A small note. It seems the locking changes uncovered several other problems with the platform in general. One example: https://github.com/openwrt/openwrt/commit/173d93ea23dabf4f08228dc2aaf70a004cf4141c

There are several more with the wifi driver which are not applicable to the GnuBee.

Good to see this improved!

neilbrown, 2018-04-18:
BTW, the tftpboot command doesn't need the magic number 80200000.
That is the default load address for tftpboot, but also the default address for bootm, so `tftpboot;bootm` is sufficient.

That openwrt patch is already in the upstream mtk eth driver, so we would have got it eventually.  It mostly just a fix for a false-positive warning though.
Certainly doesn't hurt to monitor what is happening in openwrt to see if anything might be useful - thanks.

neheb, 2018-04-18:
Btw has anyone noticed data corruption on the drives after a days worth of uptime? I've noticed this on multiple MIPS devices since kernel 4.9 on OpenWrt but have avoided reporting to the kernel folks since it could very well be due to some OpenWrt patch.

Easiest way to detect is to use btrfs as it detects silent data corruption and floods dmesg.

alethiophile, 2018-04-18:
I have seen significant data corruption on both hard drives and SD
cards. It's fairly annoying and prevents me from getting much done.

I attributed this originally to the age of the drives; I'm using
hardware that I had lying around. Haven't yet tested with known-good
stuff. If it is a problem with the Gnubee (software or hardware), it
definitely needs fixing.

Adirelle, 2018-04-18:
I have not noticed any corruption with 4.4 kernel. With 4.15, systemd-fsck complains about the ext4 FS in my mirrored logical volume (mirror LVM2) but not about the root on /dev/md1. When I run fsck on the former, it does not find nor fix any error. The disks were brand new.

neilbrown, 2018-04-19:
I've run an fio test that explicitly looks for corruption and it ran for over 24 hours and reported nothing.  This was on a recent 4.15 kernel.  I might try running another test - or several at once.
If anyone does experience data corruption, please report exact kernel version, and what clock speed the CPU is configured for - over-clocking could cause problems.  Also report details of the corruption. e.g. output of fsck.
I think you can get clock speed with "dmesg | grep sys_freq".  I have 90000000.
I think I'll open a new issue to gather reports in.

neheb, 2018-04-19:
This problem began since kernel 4.9. Maybe earlier but those kernels are EOL.

This does not just impact mt7621. OpenWrt's ar71xx platform is also affected. This indicates a MIPS specific problem. ARM and x86 are not affected.

This is a frustrating issue. At times, I have compiled firmware that seems to be working but the next one breaks.

This issue may be easier to trigger on low RAM devices. I'd have to test on my 32MB one.

Adirelle, 2018-04-23:
@neilbrown  About the modules: I am wondering about what are the absolutely required drivers for the GnuBEE and which are optional in the default config for gb1. I would like to compile the former in the kernel, and compile the latter as modules or disable them.

neheb, 2018-04-23:
An OpenWrt user discovered that network lockups may have something to do with flow control, so there's that.

Then again, we are talking about different drivers at this point...

xvybihal, 2018-04-24:
> PS: on a side note, no network lockups (thus reboot) with the current kernel for a week. This is promising.

I have just up 4 days, 22 hours, 21 minutes, but I agree, by this time it usualy presented itself with no network connection, so fa so good, except onel ittle thing:
```
gnubee ~ # dmesg | grep 'mtk_soc_eth 1e100000.ethernet eth0:' | egrep "port 4 link down| port 4 link up" | wc -l
129
```
The link is going up and down from time to time, I am not sure if that's normal, but I would not say so.

neilbrown, 2018-04-25:
> The link is going up and down from time to time, I am not sure if that's normal, but I would not say so.

Probably not normal.  The ethernet driver was in a fairly sad state when I got it - lots of work needed to make it work at all.  It is possible I broke something and certain that I didn't fix some things.  It's getting close to the top of my list of things to look at - after I get writing to flash working and streamline my process for building firmware images a bit.

Adirelle, 2018-04-25:
> Is that helpful at all?

It is. Thanks.

> The link is going up and down from time to time, I am not sure if that's normal, but I would not say so.

The link seems pretty stable there. It only disconnected two times a week ago, and I might be for a real cause (like myself unplugging then plugging the cable, I cannot remember).

neilbrown, 2018-04-27:
I've pushed out an update to gnubee/v4.15 which:
- fixes the data corruption bug (the network bug was already fixed)
- fixes writes to flash
- uses the configured MAC address for the network, instead of a random one (you need to rebuild the initramfs for this).
- updates to latest stable: 4.15.18

Now that writing to flash works, you can install new firmware with

    flashcp -v gnubee.bin /dev/mtd3

providing you have mtd-utils installed.

Adirelle, 2018-04-30:
Built, tested, written to flash using flashcp, rebooted. Everything seems fine.

Adonai, 2018-04-30:
Trying to build it on GB-1 itself: No luck, lzma says it can't allocate memory, 512 MiB seems not sufficient for the task

Trying to build it by cross-compiling: all good, extracted initramfs, compiled binutils, gcc-7.2.0, kernel, installed modules into initramfs, made uimage, flashed with USB stick - hmm, not booting.

Maybe it's my rootfs is to blame? It's the same one I used with 3.10 kernel.

neilbrown, 2018-05-01:
> hmm, not booting.

What exactly does this mean?  Do you have a serial console and see no messages, or do you not have a serial console and see no flashing lights, or is it no serial console and you cannot ping/connect?
If the evidence of not booting is a lack of network connection, then it is probably that  network device names have changed.  In think with 3.10 you have "eth0.1" and "eth0.2" or something like that.  With mainline you just have "eth0".  If you can plug the uSD card into a different computer and edit /etc/network/interfaces and make that change, it might be worth another try.

Adonai, 2018-05-01:
@neilbrown I don't yet have USB_2_UART to verify.
I guess you're right about network, will try to boot again today.

Adonai, 2018-05-01:
@neilbrown you were right, it was network.
Works like a charm.

neilbrown, 2018-05-08:
> I tried to run rsnapshot to create a recent backup of my desktop, but rsync ended up getting OOM killed,

Interesting... some sort of kernel memory leak I guess.
If you can trigger it again, it would be good if you could provide the contents of /proc/slabinfo when the machine is OOM.  I might not need that, but it will probably help.
Thanks.

hvontres, 2018-05-09:
I tried to reproduce the issue tonight, but the job ran fine. I am guessing that the combination of network traffic/ disk activity might have caused the memory leak. Please let me know if there is anything else I can do the help diagnose this.
The disk in question is a raid10 array with an LVM volume on top of it.

neheb, 2018-05-19:
Does valgrind work on mainline? In OpenWrt i'm getting:

==1804== Invalid read of size 4
==1804==    at 0x48C0ED4: free (in /usr/lib/valgrind/vgpreload_memcheck-mips32-linux.so)
==1804==    by 0x40865B4: ??? (in /lib/libc.so)
==1804==  Address 0x1e0d8 is not stack'd, malloc'd or (recently) free'd

On the topic of memory leaks, downloading a torrent with transmission ends up causing the kernel to kill it unless I add a swap device. It also seems that memory leaked by transmission doesn't seem to be freed as I still get OOM errors when logging on with ssh after disabling swap. Not sure if valid in mainline.

neilbrown, 2018-05-19:
> Does valgrind work on mainline?
Nope.  I try "valgrind ls -l" and get a segmentation fault.
The same thing happens on my 4.4.131 kernel.  Maybe some needed CONFIG option is missing.

> On the topic of memory leaks,...
I'm not surprised that transmission gets killed - it probably uses too much memory.  Probably safest to leave swap enabled I guess...  You could look in /proc/meminfo and /proc/slabinfo to see if there is any sign of excess memory usage ... though they probably take a bit of expertise to understand them fully.

neheb, 2018-05-19:
One thing I noticed is that top lists a single transmission-daemon instance whereas htop shows 4.

/proc does not show the extra three that get listed by htop. I wonder if limiting the CPUs to 1 will fix this...

neilbrown, 2018-05-20:
htop is probably showing threads while top is showing processes.
If you have the pid as $PID, then look in
  /proc/$PID/task
There will be one directory for each thread.

NeilBrown

xvybihal, 2018-05-20:
This seems to be transmission daemon problem:
https://github.com/transmission/transmission/issues/313

neheb, 2018-05-20:
Yeah I looked at the code. It's threads. Seems this is a known issue...

neheb, 2018-05-20:
I just did some testing in OpenWrt. Apparently in a default fashion, iperf results with the ethernet driver there give very inconsistent results. However, if I change the default smp_affinity from "f" to "8", there's a massive improvement in both consistency and performance. I have no idea if this can be done on a driver level.

neilbrown, 2018-05-20:
Oooh, that's interesting. It's probable that something can be done in
the driver.  The first step would be understanding how the driver works
- I don't think that just hard-coding the affinity would be welcome.
So we are probably still a fair way from a proper solution, but this is
a valuable contribution towards it.

Thanks,
NeilBrown

On Sun, May 20 2018, Rosen Penev wrote:

> I just did some testing in OpenWrt. Apparently in a default fashion, iperf results with the ethernet driver there give very inconsistent results. However, if I change the default smp_affinity from [...]

neheb, 2018-05-20:
I additionally did testing with nr-cpus=1 and nr-cpus=2. nr-cpus=1 gave me the most consistent results of ~540mbps +-10mbps. nr-cpus=2 gives me the same results. 

BUT, setting smp_affinity of the ethernet IRQ to either 2 or 4 (I forgot) such that /proc/interrupts show the IRQ being active only on CPU1 boost performance to around ~800mbps +- 10mbps.

I assume nr-cpus=2 limits mt7621 to the physical cores and keeps the SMT cores away.

Removing nr-cpus completely and setting smp_affinity to 8 gives me ~930mbps in the best case. I say best since that's around the limit of gigabit ethernet on iperf. I also say best as the variance comes back, but not as badly as before. ~770mbps IIRC was the worst case.

Without tweaking smp_affinity, iperf results go down to ~440mbps as as high as ~900mbps which is quite ridiculous.

andreygursky, 2018-05-23:
@xvybihal:
> This seems to be transmission daemon problem:

For the issue with transmission I've found a workaround and described it here: https://github.com/transmission/transmission/issues/313#issuecomment-332543087.

This is not a memory leak.

neilbrown, 2018-05-31:
Submitting patches is documented in the kernel tree, in Documentation/process/submitting-patches.rst.
It doesn't explicitly mention staging, but the same process applies.
Documentation/process/2.Process.rst has a bit about drivers/staging.
Prepare a patch and email to Greg Kroah-Hartman <gregkh@linuxfoundation.org>  and devel@driverdev.osuosl.org  and me (I'm listed in the TODO file).
get_maintainers.pl might give you a few too many addresses - definitely don't send to  John Crispin, he has said he isn't interested.  If you want you could just send a patch to me (neil@brown.name) and I'll tell you if you did anything seriously wrong, and exactly where it should be sent.

xvybihal, 2018-08-01:
After some reboots lately, I noticed that my GB1 is getting different IPs from my DHCP instead of reserved lease. After little investigation I noticed that at every reboot the GB1 has new MAC. Is that by design? Why?

From kernel log:
`mtk_soc_eth 1e100000.ethernet: generated random MAC address d2:44:2f:d4:75:c1`

neilbrown, 2018-08-01:
> After little investigation I noticed that at every reboot the GB1 has new MAC. Is that by design? Why?

The gnubee network hardware doesn't have a MAC address burnt into to, so one has to be loaded by software.
The libreCMC kernel (and probably others) have a hack in the kernel to read a mac address out of the flash memory and use that.  The devicetree file contains:

> &ethernet {
        mtd-mac-address = <&factory 0xe000>;
};

which says "read 6 bytes from offset 0xE000 in the "factory" partition of the flash, and use that for the mac address.
The upstream kernels don't have this.  However the initramfs that I provided with the kernels that I provide contains:

> addr=\`dd 2> /dev/null if=/dev/mtd2 bs=1 skip=57344 count=6 | od -t x1 -A none\`
> ip link set eth0 address \`echo $addr | tr ' ' :\`

in the init script.  This reads the same 6 bytes out of the same place in flash, and uses them to set the mac address for eth0.
So: what kernel are you running?  And what do you get if you run

> dd 2> /dev/null if=/dev/mtd2 bs=1 skip=57344 count=6 | od -t x1 -A none

??

[68 more replies]