gnubee-git/GnuBee_Docs #77: Debrick after erasing flash

leocrawford, 2018-04-04:
I stupidly issued an 

    erase all

command on the uboot terminal. This has cleared the uboot image. I can see some debrick images in this project, but can find no instructions. Are you able to help?

pinney, 2018-04-04:
OK so the bootloader is now gone...OUCH.

Like many things on these devices there are more than a few ways to get where you want to go.

Some Options you may like better than others...

Here you can find one toy used to fix a broken toy :
http://vonger.cn/?p=8054
Although I haven't used this method the idea seems sound.
It also seems one may be able use any device with a MT762x SoC.  

Desolder the SPI Flash Chip and use a ch341a (or other) programmer.

Put the desoldered ship on a SOP to DIP adapter and install a DIP8 socket.

A search for 'OpenWrt debrick' will have many hits such as :

https://wiki.openwrt.org/doc/howto/generic.debrick

https://forum.openwrt.org/viewtopic.php?id=54717

https://forum.openwrt.org/viewtopic.php?id=28343

leocrawford, 2018-04-05:
Thanks. Looks like I'm going to have some fun.

When it comes to writing back the flash image, am I right in saying that it's just the uboot partition I need to restore? There are three uboot repos on the gitgub page. Which do I need? Is there an image I can use rather than building from source?

Presumably I will have lost the flash partitions. I can find the list (u-boot, u-boot-env, factory,firmware,kernel,rootfs,rootfs_data) but can'd a record of their positions on flash. Is that available? Or do I just need to know the relative position of uboot on the flash to write it back?

Sorry for all the questions.

leocrawford, 2018-04-08:
Thanks for the links.  This is unfamiliar territory for me, but as I don't fancy testing my soldering skills the easiest option is write to the flash in-place using SPI.

As I already have a raspberry PI it appears I can do it with this:

https://www.flashrom.org/RaspberryPi

I'll be using the pin-out shown here:

https://github.com/gnubee-git/GnuBee_Docs/blob/master/DeBrick/debrick-02.png

When it comes to writing back to the flash it seems (https://wiki.openwrt.org/doc/techref/flash.layout) that the relative locations of the partitions is encoded in uboot and/or the kernel, i.e. there is no partition table to worry about.

I assume I therefore write uboot to location 0. However it is unclear to me where any other partitions would sit, and indeed if they are needed. 

My best guess (from the debian logs) is the following:

[    3.652000] 0x000000000000-0x000000030000 : "Bootloader"
[    3.664000] 0x000000030000-0x000000040000 : "Config"
[    3.676000] 0x000000040000-0x000000050000 : "Factory"
[    3.688000] 0x000000050000-0x000031050000 : "firmware"

Could you confirm this?

As I do not have backups of the partitions I will be using the binary provided above. There seem to be many versions, depending on serial speed, processor speed and RAM speeds. I assume for a normal GNU BEE 1 I need the following?

https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PCx/firmware/uBoot/115200/GB_PCx_uboot_115200-880MHz-1066RAM.bin

My assumption is that if I write this to position 0 using 'flashrom' I'll get uboot, but not 'config' or 'factory'. Is this correct? If so do you have them? Or will they be created by uboot?

Thanks in advance for your help, and providing such an interesting piece of kit.

leocrawford, 2018-04-09:
> Yes, you have the correct partition locations there (though the end of
> "firmware" is rather bizarre, it should be 2000000).
> None of the u-boot images seem to match what I have on my gnubee-pc1.
> I have a uart speed of 57600 and CPU freq of 900MHz - I don't know the
> memory speed.
> Yes, just writing u-boot to address 0 should be sufficient.
> config contains "environment variables" used by u-boot. These can all be
> set once you have a u-boot prompt, if you need them. See below for a list
> of some of mine.
> The "factory" only contains a default mac address I believe - and only
> some kernels know to look for it.
> If you can write the u-boot, you should be well on your way.
>
> My 'config' includes:
> bootcmd=tftp
> bootdelay=5
> baudrate=57600
> eth0addr=90:50:5A:01:00:09 90:50:5A:01:00:0A
> ethaddr=90:50:5A:01:00:01
> bootfile=GB-PCx_uboot.bin
> stdin=serial
> stdout=serial
> stderr=serial
> autostart=no
> ethact=eth2
> ipaddr=192.168.1.99
> serverip=192.168.1.4
> start=no
> erip=192.168.1.100

Thanks. very helpful. How did you find your CPU frequency? through software
or by hardware inspection?

neilbrown, 2018-04-09:
The main give-away for CPU freq is the message printed on the serial port at boot:
>  #### The CPU freq = 900 MHZ #### 

also when I booted a kernel that expects the freq to be 850, the real-time-clock drifted.
If I run "dmesg | grep sys_freq" it tells me "sys_freq: 90000000".
I thought there would be something in  /proc/cpuinfo, but there doesn't seem to be.
You wouldn't be able to tell by hardware inspection as the value is set by software.  The crystal provides a 40MHz clock, and a PLL (or similar) is used to multiply that up to whatever you want.  The difference u-boot versions program different multipliers.

neilbrown, 2018-05-13:
Hi, I'm really curious how you progressed here.  Did you manage to flash u-boot back again and get your gnubee working?

leocrawford, 2018-05-13:
Not yet. Had to buy a solder iron, and first attempt at soldering the pins
onto the pi didn't go well. Not ready to graduate to soldering onto the
bee. Need to find a friend to do it.

wdj2005, 2018-05-26:
Is it necessary to desolder the Flash chip? 
The schematic seems to suggest that the chip’s Write Protect pin is held high - which presumably means the chip can not be written to while its in situ. 
Which also raises the question as to how it could have got erased, if the write protect pin is held high.

neilbrown, 2018-05-26:
> The schematic seems to suggest that the chip’s Write Protect pin is held high

You are presumably looking at the SPI ROM inset on page 2.  The WP pin, as well as a few others, is marked with a # (i.e. WP#). I think that means the pin is active-low.  i.e. If the WP is tied to ground, then you cannot write.  If it is tied to 3.3V, then it isn't write-protected.

wdj2005, 2018-05-27:
Ahhh. Maybe it will be ok. I will leave Leo to wire the Raspberry Pi to it and try flashrom

wdj2005, 2018-05-27:
Flashrom detects the chip, but claims it’s not a supported device. :-(
[image]

pinney, 2018-05-27:
./flashrom -p ch341a_spi
flashrom p1.0-22-g0bfa819 on Linux 4.9.0-4-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
Found Winbond flash chip "W25Q256.V" (32768 kB, SPI) on ch341a_spi.
No operations were specified.

leocrawford, 2018-06-22:
Wow! It's all worked, exactly as described above. I flashed using:

flashrom p1.0-77-gf9a3055 on Linux 4.9.80+ (armv6l)

compiled form source on a raspberry pi, writing the u-boot image directly to the rom (I had to use dd to pad with zero's to make it the right length) 

I'm currently getting the following errors, but I assume I just need to install a kernel and all will be well. fingers crossed!

```
Bad Magic Number,00000000 
 
## Enter to Rescue Mode (image error) ##

 NetTxPacket = 0x9BFB6CC0 

 KSEG1ADDR(NetTxPacket) = 0xBBFB6CC0 

 NetLoop,call eth_halt ! 

 NetLoop,call eth_init ! 
Trying eth2

 ETH_STATE_ACTIVE!! 
Using eth2 device

Our IP address is:(192.168.1.1)
Wait for TFTP request...
```
Thanks so much to everyone for their help.

leocrawford, 2018-06-24:
Having installed a kernel I now get pages of the following error:

```
[   14.676239] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050000: 0x5847 instead
[   14.695536] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050004: 0xbea8 instead
[   14.714828] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050008: 0xaba5 instead
[   14.734120] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0005000c: 0x473b instead
[   14.753415] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050010: 0xaeda instead
[   14.772707] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050014: 0x221a instead
[   14.791999] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050018: 0xb135 instead
[   14.811292] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0005001c: 0xe703 instead
[   14.830582] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050020: 0xa414 instead
[   14.849871] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00050024: 0xcd19 instead

```
The kernel is the standard librecmc one from the website, and is installed using USB.

Any ideas?

neilbrown, 2018-06-25:
> The kernel is the standard librecmc one from the website, and is installed using USB.

Which librecmc kernel exactly?  It can help to be explicit.
I think that firmware mounts /dev/mtdblock6 from the internal flash as a jffs2 filesystem at /overlay/upper.
Then it uses overlayfs to mount that over a squashfs root image that is at /dev/mtdblock5.

The jffs2 filesystem starts at 16056320 (from /sys/class/mtd/mtd6/offset)
So if the image that you wrote is smaller than that (like librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin which is only 9961638) then it won't contain a jffs2 filesystem.
If your image is larger than that (like GB-PCx/firmware/outdated/gnubee-shipped.bin) then it will have a jffs2 filesystem in it.
I think the idea is that when you flash a 'sysupgrade' firmware, you don't lose what is stored in the jffs2 filesystem.
I suggest you flash the gnubee-shipped.bin firmwhere, then flash again wit hthe sysupgrade firmware of your choice.

pinney, 2018-06-25:
The older gnubee-shipped.bin firmware may fix this problem however you should also run into another problem since you 'started over' with a blank flash chip.
Namely the 'factory' partition is also blank...It contains the MAC address.
However the device should boot and run with a random MAC address which will of course be randomized again at boot up.

leocrawford, 2018-06-26:
It looks like it was the choice of firmware that cause the problem. I flashed with gnubee-shipped.bin and it all worked. I'll now flash the factory partition back to the flash. Thanks so much for all the help.

closed 2018-06-26 by leocrawford