gnubee-git/GnuBee_Docs #79: ffmpeg execution time with 4.15 kernel

Adirelle, 2018-04-22:
With Neil Brown's mainline kernel, ffmpeg is now running -- whereas I had "illegal instruction error" with LibreCMC one's -- but is has terrible performance. See below : asking for --help takes around 46 seconds... with 40s for the system.

Has anyone an idea of what is happening and a workaround ? Should I try to recompile it with the cross-compiler used to compile the kernel ?

```
# cat /proc/version
Linux version 4.15.17+ (guillaume@localhost) (gcc version 7.2.0 (GCC)) #11 SMP Mon Apr 16 19:10:25 CEST 2018

# time ffmpeg --help
ffmpeg version 3.2.10-1~deb9u1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Debian 6.3.0-18) 20170516
  configuration: --prefix=/usr --extra-version='1~deb9u1' --toolchain=hardened --libdir=/usr/lib/mipsel-linux-gnu --incdir=/usr/include/mipsel-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --disable-mipsdsp --disable-mipsdspr2 --disable-loongson3 --disable-mips32r6 --disable-mips64r6 --disable-mipsfpu --enable-shared
  WARNING: library configuration mismatch
  avutil      configuration: --prefix=/usr --extra-version=2 --toolchain=hardened --libdir=/usr/lib/mipsel-linux-gnu --incdir=/usr/include/mipsel-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --disable-mipsdsp --disable-mipsdspr2 --disable-loongson2 --disable-loongson3 --disable-mmi --disable-mips32r5 --disable-mips32r6 --disable-mips64r6 --disable-msa --disable-mipsfpu --enable-shared
  avcodec     configuration: --prefix=/usr --extra-version=2 --toolchain=hardened --libdir=/usr/lib/mipsel-linux-gnu --incdir=/usr/include/mipsel-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --disable-mipsdsp --disable-mipsdspr2 --disable-loongson2 --disable-loongson3 --disable-mmi --disable-mips32r5 --disable-mips32r6 --disable-mips64r6 --disable-msa --disable-mipsfpu --enable-shared
  swresample  configuration: --prefix=/usr --extra-version=2 --toolchain=hardened --libdir=/usr/lib/mipsel-linux-gnu --incdir=/usr/include/mipsel-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --disable-mipsdsp --disable-mipsdspr2 --disable-loongson2 --disable-loongson3 --disable-mmi --disable-mips32r5 --disable-mips32r6 --disable-mips64r6 --disable-msa --disable-mipsfpu --enable-shared
  libavutil      55. 34.101 / 55. 58.100
  libavcodec     57. 64.101 / 57. 89.100
  libavformat    57. 56.101 / 57. 56.101
  libavdevice    57.  1.100 / 57.  1.100
  libavfilter     6. 65.100 /  6. 65.100
  libavresample   3.  1.  0 /  3.  1.  0
  libswscale      4.  2.100 /  4.  2.100
  libswresample   2.  3.100 /  2.  7.100
  libpostproc    54.  1.100 / 54.  1.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

[... snip ... description of options ... snip ...]

real    0m46.365s
user    0m4.390s
sys     0m40.040s
```

neilbrown, 2018-04-22:
If I run "gdb ffmpeg" and give the command "run --help", then wait a while and interrupt with ctrl-C, then "where" will tell me what code is running at that moment.
The "c" for continue, wait and bit more and crtl-C and "where" again and I get another snapshopt.

I looks like all the time is spent doing initialization for /usr/lib/mipsel-linux-gnu/libopenmpt.so.0.  openmpt is an audio decode.
The mt7621 has no floating point support, so anything that needs lots of numerical calculations is likely to be quite slow.

Adirelle, 2018-04-23:
This is what I was a bit afraid of. Thanks for you analysis.