gnubee-git/GnuBee_Docs #80: Can't build openwrt image, http 404

Adonai, 2018-04-29:
```
make[2]: Leaving directory '/gnubee-openwrt/toolchain/gdb'
make[2]: Entering directory '/gnubee-openwrt/toolchain/kernel-headers'
mkdir -p /gnubee-openwrt/dl
/gnubee-openwrt/scripts/download.pl "/gnubee-openwrt/dl" "linux-3.10.14-p112871.tar.gz" "x" "https://mqmaker.com/git-download.php?user=mqmaker&proj=linux&archive="
converted 'https://mqmaker.com/git-download.php?user=mqmaker&proj=linux&archive=/linux-3.10.14-p112871.tar.gz' (ANSI_X3.4-1968) -> 'https://mqmaker.com/git-download.php?user=mqmaker&proj=linux&archive=/linux-3.10.14-p112871.tar.gz' (UTF-8)
--2018-04-29 12:34:08--  https://mqmaker.com/git-download.php?user=mqmaker&proj=linux&archive=/linux-3.10.14-p112871.tar.gz
Resolving mqmaker.com (mqmaker.com)... 137.59.148.71
Connecting to mqmaker.com (mqmaker.com)|137.59.148.71|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-04-29 12:34:11 ERROR 404: Not Found.

Download failed.
converted 'http://mirror2.openwrt.org/sources/linux-3.10.14-p112871.tar.gz' (ANSI_X3.4-1968) -> 'http://mirror2.openwrt.org/sources/linux-3.10.14-p112871.tar.gz' (UTF-8)
--2018-04-29 12:34:11--  http://mirror2.openwrt.org/sources/linux-3.10.14-p112871.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-04-29 12:34:11 ERROR 404: Not Found.

Download failed.
converted 'http://downloads.openwrt.org/sources/linux-3.10.14-p112871.tar.gz' (ANSI_X3.4-1968) -> 'http://downloads.openwrt.org/sources/linux-3.10.14-p112871.tar.gz' (UTF-8)
--2018-04-29 12:34:11--  http://downloads.openwrt.org/sources/linux-3.10.14-p112871.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 148.251.78.235, 2a01:4f8:202:43ea::3
Connecting to downloads.openwrt.org (downloads.openwrt.org)|148.251.78.235|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://sources.openwrt.org/linux-3.10.14-p112871.tar.gz [following]
converted 'http://sources.openwrt.org/linux-3.10.14-p112871.tar.gz' (ANSI_X3.4-1968) -> 'http://sources.openwrt.org/linux-3.10.14-p112871.tar.gz' (UTF-8)
--2018-04-29 12:34:11--  http://sources.openwrt.org/linux-3.10.14-p112871.tar.gz
Resolving sources.openwrt.org (sources.openwrt.org)... 148.251.78.235, 2a01:4f8:202:43ea::3
Reusing existing connection to downloads.openwrt.org:80.
HTTP request sent, awaiting response... 404 Not Found
2018-04-29 12:34:12 ERROR 404: Not Found.

```

pinney, 2018-04-29:
cd gnubee-openwrt/dl
wget https://github.com/mqmaker/linux/archive/3.10.14-p112871.tar.gz

Adonai, 2018-04-29:
Thanks! Should the build git be updated with working link?

neilbrown, 2018-04-30:
@Adonai if you create a pull request with the change, I suspect there is an excellent change it would be merged.

Adonai, 2018-04-30:
@neilbrown , good idea but download is being done by quering mirrors, they are handled in `scripts/download.pl`. On each mirror the same filename is expected to be present and it's "linux-3.10.14-p112871.tar.gz", so I can't solve this by doing
```diff
push @mirrors, 'http://mirror2.openwrt.org/sources';   
push @mirrors, 'http://downloads.openwrt.org/sources'; 
+push @mirrors, ' https://github.com/mqmaker/linux/archive/'; 
```

neilbrown, 2018-05-01:
> so I can't solve this by doing

Surprisingly you can (if you drop the space before https).  I started hacking the script to replace the name, messed it up, but it still worked.  I changed to just add the 'push' line, and it still worked.  I guess github is clever.

Adonai, 2018-05-01:
@neilbrown I didn't think so much help and knowledge could come from a single human being.
I rased PR now, it's [#4](https://github.com/gnubee-git/gnubee-openwrt/pull/4)

closed 2018-05-01 by Adonai