gnubee-git/GnuBee_Docs #81: How to build just one module for 3.10.14 kernel?

Adonai, 2018-04-29:
I have a setup where whole GB-1 can be controlled from root image on a microSD. I need tun driver for it and wanted to build it from source, but neither `/proc/config.gz` nor raw config image from mqmaker source tree don't seem to match the 3.10.14 kernel that GB-1 has.

How to build just one module for the mqmaker kernel that can be loaded into running system?

Adonai, 2018-04-30:
No, no, that's not what I'm saying :) /proc/config.gz is simply nonexistent on 3.10

neilbrown, 2018-04-30:
Ahh - I misunderstood.  So does the running kernel appear to match https://github.com/gnubee-git/gnubee-openwrt/blob/master/target/linux/ramips/mt7621/config-3.10
??

Adonai, 2018-04-30:
... I wish I had that link earlier. I'll try to build a module now and see how it goes.

Adonai, 2018-04-30:
@neilbrown  looks like it's not. Had to enable loadable kernel support by hand, set CONFIG_HZ and PREEMPT and some other options.
Will try to build module with this config nevertheless.

To clarify, I have the system from this page: http://gnubee.org/install_debian.html, first link

neilbrown, 2018-04-30:
That's unfortunate.
If you pass the Code: line as input to ./scripts/decodecode it will show you some code.
If you make drivers/net/tun.s it will create an assembly language version of the module.
If you find the match, you will see it is in tun_net_init() where it calls eth_hw_addr_random().  dev->dev_addr is NULL.  This gets set by dev_addr_init().  I don't think that can possibly have not happened.
So there must be some difference in the configs which results in 'struct net_device' having fields in different positions, possibly  CONFIG_WIRELESS_EXT for example.  Actually it was probably CONFIG_NET_NS.
You really need to find exactly the same kernel source and config that was used to build the kernel you are using.
Try this config: https://github.com/gnubee-git/gnubee-openwrt/blob/0a213723e9d43411a35eb68d00b64f64095d7765/target/linux/ramips/mt7621/config-3.10 
it is from closer to the build time of that kernel.
I think the kernel was built in gnebee-openwrt which might contain patches on the mqmaker kernel.

Adonai, 2018-04-30:
@neilbrown , no luck, same error.
There's not much we can do here. Maybe GnuBee guys could tell which config did they use, so I'll leave this issue open for a week or two.

 Meanwhile I think I'll take a look at mainline kernel you compiled earlier.