gnubee-git/GnuBee_Docs #82: GB-PC2 Schematic Seems to have errors

hvontres, 2018-05-14:
I noticed a few discrepancies between the GB-PC2 schematic pdf and the actual hardware:

1. 1. Sheet 5 only shows two of the three ehternet connectors
2. 2. Sheet 10 shows the power led connected to GND, but it seems to be GPIO controlled on my board
3. The title in Okular still shows as gb-pc2 v1

Thanks for all your work

neilbrown, 2019-03-27:
> I noticed a few discrepancies between the GB-PC2 schematic pdf and the actual hardware:
> 
>     1. Sheet 5 only shows two of the three ehternet connectors

Correct.  Sheet 17 shows the third.  It is connected to a separate PHY, rather than one of the phys integrated into the SOC.

> 
>     2. Sheet 10 shows the power led connected to GND, but it seems to be GPIO controlled on my board

Interesting.... The two leds near the power switch are connected to GPIO6 and GPIO8, neither of which are shown on the schematic.  This is correct on the GB-PC1 schematic.
GPIO24 (shown on sheet 10) is connect to the LED on the black ethernet port on the PC1, but doesn't seem to do anything on the PC2.  Both Schematics show the green Ethernet LEDs for the first 2 ports non connected.  This might be correct on the PC2.
The PC2 schematics suggests the two LEDs on the first port are connected to GPIOs 13 and 15, but I cannot make them work by toggling those GPIOs.

Strange.

Thanks for mentioning this.

> 
>     3. The title in Okular still shows as gb-pc2 v1
> 
> 
> Thanks for all your work

jjakob, 2020-01-07:
Reverse engineering my GB-PC2 V1.1:

- The blue and black port green LEDs **are** connected to the SoC to **I2C** SCLK and SD. They have unpopulated pull-ups R1609/R1610.
- The yellow port orange led is connected to R69 and R66 so that does seem to go to GPIO13/JTDO. I can't see where the green LED goes as the track goes under the heatsink. I can't see R65 outside so that's likely hidden under the heatsink as well which means it probably equates to the schematic as well.
- The front green system LEDs differ from the schematic and are the same as on PC1 (they already work/can be controlled by gb-pc1:green:system and status)

jjakob, 2020-01-07:
GPIO24/25 map to pins Y17/Y18 which are GE2_TXD3/2 for RGMII IP1001 for yellow ethernet port. These aren't switched to GPIO mode so the lan1/lan2 definitions can be removed. The lan3 needs to be added with GPIO13 and 15.

neilbrown, 2020-01-19:
I can make the green LEDs on the Black and Blue ports work by requesting GPIOs 3 and 4 (I2C SD and SCLK), but I still cannot control either LED on the yellow port.
They are both always ON for me.  I've tried configuring and toggling GPIOs 13-17, but they have no visible effect.
So we have some progress, but we aren't there yet.
Thanks for your research.

jjakob, 2020-01-19:
> I can make the green LEDs on the Black and Blue ports work by requesting GPIOs 3 and 4 (I2C SD and SCLK), but I still cannot control either LED on the yellow port.
That's great.
> They are both always ON for me. I've tried configuring and toggling GPIOs 13-17, but they have no visible effect.
Perhaps some register is still set wrong, both are always **off** for me so you do seem to be getting them to do something. They're connected between 3.3V and the pin so the pin state is active low (inverted) as all the other LEDs.

neilbrown, 2020-01-25:
There are new 5.4.14 firmware images on http://neil.brown.name/gnubee.  The pc2 version has code to control GPIOs 13 and 15 using /sys/class/leds/gb-pc2:green:lan3 and gb-pc2:orange:lan3.  They don't work for my.  I'd appreciate it if you could try it out and see what the LEDs do for you.

bornlibra23, 2020-04-30:
Hi could you give a link to the schematic? I cant find it anywhere. I am trying to upgrade the number of sata ports.

bornlibra23, 2020-04-30:
Can you publish the updated schematic?

jjakob, 2020-05-01:
I don't think anyone has the raw schematic we could update, and I'm certainly not going to redraw it from the PDF. I could draw just the differences I've found, but I don't have the time right now.