gnubee-git/GnuBee_Docs #86: Typical performance values?

nariox, 2018-08-08:
Hello all,
I think it'd be helpful for users to have an idea of what typical performance we can get out of the GB-PCxs.

I remember when I first got my GB-PC1, I could get 100MiB/s+ on hdparm, but now (for some unknown reason) I get stuck at around ~80MiB/s. I know my drives can go higher than that (one of them is a SSD), but can't figure out if there's a problem with my unit or if the number I got initially were not accurate.

Also, I've set up a Time Machine server (using `samba 4.8.2`) for my wife. There, I'm get about 40MiB/s (connected through gigabit ethernet). Not sure if those values are typical, but the MT7621A cpu goes to 100% when I'm testing it. I had used `netatalk 3.1.1` previously, but the performance was even worse (although it might be the AFS implementation I had).

Could you guys share what performance values you are getting so I can get an idea whether I'm doing something wrong?

anarcat, 2019-01-26:
i've opened a separate issue to figure out the exact specs of the SATA controller, which are a little hazy as well. It shouldn't be the bottleneck here as the data sheet says it should do 6Gbps, but you never know...

In any case, if you don't max your network (e.g. assuming gigabit), the next issue you will have is likely to be disks. Only SSDs and most recent HDDs can max a gigabit link. Once you cross that mark, of course, CPU will be a problem, but then hdparm should be able to bypass *some* of those bottlenecks at least, and you will definitely see CPU usage if that becomes a bottleneck.

Considering the fairly slow clock speed, I would certainly expect the CPU to be a bottleneck for some operations.

neheb, 2019-01-27:
A bunch of russians overclocked a XiaoMi 3G to 1100Mhz (same CPU) by taking off a resistor. I tried looking for it on the PC1 Circuit board but couldn’t find it.

nariox, 2019-01-27:
Well, even though the ethernet is gigabit, doesn't mean the switch is actually able of doing sustained 1gpps transfers. But since this processor is used in routers, I would guess at least the network is capable of higher throughput (also iperf/netperf confirm it).

@neheb : this would probably only work if you flashed a custom U-boot. The default one is not capable of overclocking.