gnubee-git/GnuBee_Docs #87: UART speed

vgiralt, 2018-08-31:
I do not have a compatible USB-2_UART dongle, but I _do_ have a machine with a real RS-232 DB9 port. I have connected the TX RX and GND pins to the corresponding DB9 pins, but I only get garbage at any speed :-( Does anyone know the configured speed for a direct RS232 connection?
I need to peek into the boot sequence as I cannot get Debian to boot of an SD card despite the image I use :-(
Thanks.

vgiralt, 2018-08-31:
Thanks for the response. That's the first configuration I tried, I had found the documentation. I'm using @neilbrown gnubee-pc1-900-4.4.144.bin image. Do any of you know if U-Boot needs to receive some combination of characters to set de speed or something like that?

pinney, 2018-09-01:
https://openwrt.org/docs/techref/hardware/port.serial

Most common cause of a 'Garbled Output" is a bad ground connection.

Have you tested your RS-232 connection?

vgiralt, 2018-09-02:
Thanks again @pinney! 
The RS-232 was working. I also use it for the console of a CISCO router. Although I could not test the ground issue, as I had no better way to connect the pins on the GnuBee than the horrible contraption I created using some PC internal wires I had laying around in a drawer.
You are right with the voltage risks. Although RS-232 levels can be anywhere from 3 to 25 volts (plus or minus) the tend to be around 12. It's been ages since I used to fiddle with serial connections ...
Anyhow, I got the thing working thanks to something I read in some comment and that I had read on the docs, but simply forgot: **modern kernels do not boot off SD cards**. Man, I have a six 1TB disk array, Why insist on having the OS on SD?
OFF-TOPIC: problem was that it _seemed_ to be booting, as the network configuration was applied and _some_ of the commands (mostly pings) in an if-up-post.d script where executed.
My excuses for creating noise.

closed 2018-09-02 by vgiralt