gnubee-git/GnuBee_Docs #92: GnuBee1 unresponsive after plugging USB DVD reader

vgiralt, 2019-01-14:
I've plugged a CD/DVD reader/writer in the front (blue) USB port of my GnuBee1 and now it is simply frozen, the LEDs how a very dim light, no flashing, and there are no life signs over the serial port, just noise.
The six disks are humming, but no booting.
Ideas anyone?
Thanks!

nariox, 2019-01-15:
Have you tried with a different power supply? One thing to try would be removing the HDDs and seeing if the serial port shows some sign of life with a good power supply and no HDDs connected.

vgiralt, 2019-01-15:
Thanks Nariox. I thought about that, but had no power supply at had. I left the GnuBee to rest overnight and it's working.
So, let's leave this here for posterity and close the issue,

closed 2019-01-15 by vgiralt