gnubee-git/GnuBee_Docs #93: Reccomend against connecting to the network before changing the default password

Qwertie-, 2019-01-22:
The current debian install guide informs the user to plug the gnubee in to the network, ssh in and then change the password. In the age of automated scanning of devices with default passwords, its entirely possible the device has been found and already compromised by this point. The guide should instead direct the user on how to configure a password before the device has connected to the network. Either by directly connecting the gnu bee via ethernet or by changing the install script to ask for a password before install.

Cabalist, 2019-01-22:
Correct me if I am wrong here but automated scanning should not be a concern on your local network.

 The only time this would be an issue is if you are immediately exposing the GnuBee to the internet without a firewall or setting it up in an environment that you cannot trust (i.e. a internet cafe or guest network).

Qwertie-, 2019-01-23:
@Cabalist I guess it depends how your specific network is set up but I would consider any device that exposes passwordless/default password access to anything that can access it to be insecure. Its possible you may have no NAT if you are on ipv6 or that one of the devices on your local network is compromised and is scanning for other devices.

In any case its still a very good idea to have a password set up before exposing ssh even to just the local network.

xvybihal, 2019-01-23:
Suggest the guide edit if you think it's needed. But I personaly think it's not.

You can modify debian-jessie-install script before running it and change pw there. Also, you can just use UART and change password without network.

Also, if you plug device into network and it immediately exposes it's ports to internet, you are probably doing something wrong on your network layer.

One could suggest tons of security things into the guide, but it should be lightweight starting point for new users, if you ask me.