gnubee-git/GnuBee_Docs #95: dropbear high CPU usage with rsync/scp running openwrt 18.06.1

cbm755, 2019-02-01:
I can't get above 3.5MiB/s, even with all-wired connections.  This is a GnuBee PC2 wired to router, Fedora laptop wired to router.  Is this a fundamental limitation of this CPU or or some hardware acceleration missing?

```
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1878  1242 root     R     1160   0%  25% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 1880  1879 root     S     3304   1%   7% rsync --server -vulogDtpre.iLsfxC --delete . /mnt/files/tmp/
```
If start multiple rsync jobs, I can get close to 3MiB/s for each of them and then there are multiple 25% dropbear processes.  So its CPU bound, no disk/network IO.  (Note "25%" here means, I think, 100% on one core).

This is a out-of-box PC2 flashed to OpenWRT 18.06.1.

neheb, 2019-02-01:
Slow CPU. Either force outdated crypto if you can or do not use rsync.

There is a driver for accelerating AES but CTR mode (which ssh uses) has not been implemented yet. The driver so far has been able to get ~40MB/s on AES-CBC (tested with cryptsetup benchmark).

One workaround that I used to use was to setup a Samba share, mount it with mount.cifs, and rsync on that directory. Not perfect (has issues with Unicode) but faster. Just make sure to disable sendfile in Samba as that will panic the kernel after a while).

cbm755, 2019-02-01:
Thanks!  I've never known anything but rsync/ssh so will have to look into alternatives :-)  `rsh` maybe or FTP server for initial transfer then regular rsync for incremental backups...

- - - - 

I tried `scp -c chacha20 ...`, `scp -c arcfour ...` and a few others but so far I always get:
```
Unknown cipher type 'arcfour'
```
Searching suggests I may need to reconfigure (or recompile openssh).  Do you have a command line example that works?   Might be good to add this to the FAQ wiki entry...

neheb, 2019-02-01:
That would be chacha20-poly1305@openssh.com , not chacha20.

arcfour is compile-time disabled, as it is insecure.

Previously I've forced rsync to use chacha20 by editing ssh_config and forcing chacha20 on specific Hosts.

cbm755, 2019-02-01:
Thanks!  Still needs some config
```
Unable to negotiate with 172.16.17.200 port 22: no matching cipher found. Their offer: aes128-ctr,aes256-ctr
lost connection
```
Anyway, I have enough information now to play with it..., thanks.

I also found https://serverfault.com/questions/116875/how-can-i-disable-encryption-on-openssh

I guess I should close this?  Some combination of "Slow-CPU" and "Waiting on upstream kernel driver implementation".

neheb, 2019-02-01:
Oh I use openssh-server as it is bigger but faster than dropbear.

Driver is here if you want to follow along: https://github.com/vschagen/mtk-eip93

cbm755, 2019-02-01:
Thank  It looks like [dropbear would require, at least, recompiling](
https://superuser.com/questions/1344681/how-can-i-configure-the-ciphers-for-dropbear-on-dd-wrt)

Good news about the driver!

cbm755, 2019-02-01:
Ok, documented on wiki, closing.

closed 2019-02-01 by cbm755