gnubee-git/GnuBee_Docs #96: Not seeing anything on usb uart

Qwertie-, 2019-02-07:
I got a usb uart cable

```
  idVendor           0x067b Prolific Technology, Inc.
  idProduct          0x2303 PL2303 Serial Port

```

And wired it to the gnu bee like this

[image]

I then ran the command from the wiki `sudo screen /dev/ttyUSB0 57600` and then turned on the gnubee. I see a blank black screen and no output. The lights on the gnu bee flash but still nothing on my screen.

neheb, 2019-02-07:
You might need to swap TX and RX. Remember than both are reversed when looking at the cable. (transfer on the board is receive on the cable).

I actually have a different issue on my PC1. If I power it on with the cable plugged in, it doesn't even turn on (no lights but hard drives do spin) whereas without the cable it starts fine. Maybe a blown capacitor somewhere.

Qwertie-, 2019-02-07:
Ah that was exactly it. I see the output now. Thanks

closed 2019-02-07 by Qwertie-