gnubee-git/GnuBee_Docs #97: PC2 - No Network with gnubee-pc1-4.4.174.bin

nedson, 2019-02-10:
A few details:
* GnuBee PC2 loaded up with 4X4TB in Raid 10 and 2X3TB in Raid 1
* Booting from Debian from usb drive

# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian

# cat /etc/network/interfaces
auto lo
iface lo inet loopback
auto eth0.2
iface eth0.2 inet dhcp

Working with old firmware:
Linux gnubee-n1.gnubee 3.10.14 #1 SMP Wed Mar 1 10:20:40 BRT 2017 mips GNU/Linux

After flashing, I have no network connection. I tried manually setting the network using USB-UART cable, but cannot seem to get it working. Reverting back to the old firmware reenables network.

I was hoping to fix the network crashing bug that many have experienced. Network seems to intermittantly stop working. A reboot fixes the issue.

I know that @neilbrown runs a PC1. Perhaps there is something sightly different that is causing the PC2's network to fail.

neilbrown, 2019-02-10:
Thanks for testing.  Yes, I broke the network in a couple of recent release (in a couple if different ways).
Please try gnubee-pc1-4.4.174-1.bin

nedson, 2019-02-10:
Thanks!  Just flashed 174-1. Network seems to be back. I'll continue testing this version and report back any issues. 

I am curious to see if this version will cure the phantom network crash bug. I wrote a little script that will check for network and reboot if it goes down. Thank you again for your development - I hope to see Gnubee grow.

neilbrown, 2019-02-10:
Thanks for the confirmation - with that, I'll close this issue.

closed 2019-02-10 by neilbrown

nedson, 2019-02-11:
Using gnubee-pc1-4.4.174-1.bin, I am noticing these errors:

Feb 11 00:19:31 gnubee-n1 kernel: [  281.632387] RPC: fragment too large: 45284

neilbrown, 2019-02-11:
This shouldn't happen...
What number is in /proc/fs/nfsd/max_block_size on the gnubee?
What is the wsize seting for the mount on the NFS client - you should be able to see it in /proc/mounts.

The wsize needs to be less than the max_block_size.  The later defaults to 32768 on the gnubee (as it only has 512M of RAM, and the default is 1/4096 of total memory, but not bigger than 256M (I think)).
The wsize should default to a suitable number ... maybe you are over-riding it? Or maybe the defaults aren't working right.

nedson, 2019-02-11:
# cat max_block_size
32768

# cat mounts
/dev/sdg2 / ext4 rw,noatime,errors=remount-ro,data=ordered 0 0
tmpfs /lib/modules tmpfs rw,relatime 0 0
sysfs /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
devtmpfs /dev devtmpfs rw,nosuid,size=252860k,nr_inodes=60895,mode=755 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
tmpfs /run/lock tmpfs rw,nosuid,nodev,noexec,relatime,size=5120k 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd 0 0
sunrpc /run/rpc_pipefs rpc_pipefs rw,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,relatime 0 0
nfsd /proc/fs/nfsd nfsd rw,relatime 0 0
/dev/md0 /mnt/md0 ext4 rw,relatime,discard,stripe=256,data=ordered 0 0
/dev/md1 /mnt/md1 ext4 rw,relatime,discard,data=ordered 0 0
tmpfs /run/user/0 tmpfs rw,nosuid,nodev,relatime,size=51460k,mode=700 0 0

nedson, 2019-02-11:
Perhaps I need to explicitly set the rsize and wsize in /etc/fstab.

cat /etc/fstab
# UNCONFIGURED FSTAB FOR BASE SYSTEM
LABEL=GNUBEE-ROOT /  ext4  noatime,errors=remount-ro 0  1
/dev/md0 /mnt/md0 ext4 defaults,nofail,discard 0 0
/dev/md1 /mnt/md1 ext4 defaults,nofail,discard 0 0

nedson, 2019-02-11:
Perhaps I need to explicitly set the rsize and wsize in /etc/fstab:

cat /etc/fstab

LABEL=GNUBEE-ROOT /  ext4  noatime,errors=remount-ro 0  1
/dev/md0 /mnt/md0 ext4 defaults,nofail,discard 0 0
/dev/md1 /mnt/md1 ext4 defaults,nofail,discard 0 0

neilbrown, 2019-02-11:
The wsize is a setting on the NFS client, not the Gnubee which is the NFS server.
Where are you mounting the NFS filesystem?  What does /proc/mounts show on that machine?

nedson, 2019-02-11:
I am mounting the shares on several machines.
All of the machines indicate rsize=32768,wsize=32768

client machine 1
ip_address:/mnt/md0 /storage/md0 nfs rw,relatime,vers=3,rsize=32768,wsize=32768,namlen=255,hard,proto=tcp,port=2049,timeo=70,retrans=3,sec=sys,local_lock=none,addr=ip_address 0 0

client machine 2
ip_address:/mnt/md0 /home/nathan/md0 nfs4 rw,relatime,vers=4.2,rsize=32768,wsize=32768,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=ip_address,local_lock=none,addr=ip_address 0 0

client machine 3
ip_address:/mnt/md0 /home/pi/md0 nfs4 rw,relatime,vers=4.1,rsize=32768,wsize=32768,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=ip_address,local_lock=none,addr=ip_address 0 0

neilbrown, 2019-02-11:
OK.. The max_block_size is the maximum size for a complete NFS request - including RPC header, NFS header, and data.
The header sizes can vary - for a v4.1 WRITE that I can find in a packet trace I happen to have lying around, the data start at 0x11A, while the RPC header starts at 0x3A, so that i 0xE0 of header, or 224 bytes.  So a 32768 byte write needs 32992 bytes of space on the server.
Your fragment-too-large error said 45284 which is 12516 too big.  I can't explain that....

However, you could try - on the gnubee -
rpc.nfsd 0
echo 65536 > /proc/fs/nfsd/max_block_size
rpc.nfsd 4
exportfs -a

That should double the acceptable block size.

nedson, 2019-02-11:
Thanks for walking me through this. I'll give it a try. Appreciate all of the work you are doing on the GnuBee.