gnubee-git/GnuBee_Docs #98: Getting USB "uas" drives to work

cbm755, 2019-02-13:
I have a new USB harddrive.  It works fine on my Fedora laptop.

```
Bus 002 Device 004: ID 0bc2:ab24 Seagate RSS LLC Backup Plus Portable Drive
```

```
# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
```
(On my laptop, *two lines* are instead shown here; it can use either `uas` driver or `usb-storage`)

dmesg:
```
[ 1199.406926] scsi host6: uas
[ 1199.413775] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
[ 1199.430969] xhci-mtk 1e1c0000.xhci: @000000000f7c6ab0 0eaf7200 00000000 05000000 01078001
[ 1199.447604] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 2
[ 1199.464766] xhci-mtk 1e1c0000.xhci: @000000000f7c6ac0 0eaf7100 00000000 05000000 01038001
[ 1220.952449] scsi 6:0:0:0: tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN 
[ 1220.967877] scsi 6:0:0:0: tag#0 CDB: opcode=0x12 12 00 00 00 24 00
[ 1220.980603] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[ 1221.000603] xhci-mtk 1e1c0000.xhci: ep deq seg = 8e8f5300, deq ptr = ae969010
[ 1222.118326] scsi host6: uas_eh_device_reset_handler FAILED to get lock err -16
[ 1222.133043] scsi 6:0:0:0: Device offlined - not ready after error recovery
```
This is all on my GnuBee PC2, running OpenWRT 18.06.2 (`Linux gnubee 4.14.95 #0 SMP Mon Jan 28 08:54:32 2019 mips GNU/Linux`)

Current thoughts:
   * maybe its a power draw thing
   * maybe I can blacklist the `uas` driver and force it to use `usb-storage`.
  * maybe there is simply some kernel module I haven't installed (I have `opkg install kmod-usb-storage kmod-usb-storage-uas usbutils kmod-fs-vfat kmod-fs-isofs f2fs-tools kmod-fs-f2fs kmod-usb2 kmod-usb3`)
  * Simply removing `kmod-usb-storage-uas` did not work.

cbm755, 2019-02-13:
Update: some older enclosure works using `usb-storage`:

```
# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M

# lsusb
Bus 001 Device 004: ID 14cd:6116 Super Top M6116 SATA Bridge
```

neheb, 2019-02-13:
UAS is not possible. mtk-usb does not support it. You're stuck with BOT.

cbm755, 2019-02-13:
Ok, then we should figure out how to blacklist uas.  Once I figure out that I will document it under FAQ and close this.

Or, is this the sort of quirk that should be OpenWRT's responsibility to do this OOB on GnuBee hardware?  Should I file a bug upstream?

neheb, 2019-02-13:
UAS is provided by kmod-usb-storage-uas, or more directly, $(LINUX_DIR)/drivers/usb/storage/uas.ko . So removing one or the other should work.

UAS is only enabled by default on brcm2708, which is for Raspberry Pis.

cbm755, 2019-02-14:
Its not enough to just remove `uas.ko` because for some reason `mod-storage` doesn't show up (like it does on my laptop).  Will experiment more later...

cbm755, 2019-02-14:
xref with https://bugs.openwrt.org/index.php?do=details&task_id=1862

cbm755, 2019-02-14:
I have confirmed that the drive does work *without* `uas` on my Fedora x86 laptop:
```
[99349.561654] usb 1-2: new high-speed USB device number 13 using xhci_hcd
[99349.691084] usb 1-2: New USB device found, idVendor=0bc2, idProduct=ab24, bcdDevice= 1.00
[99349.691091] usb 1-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[99349.691095] usb 1-2: Product: BUP Slim BK
[99349.691099] usb 1-2: Manufacturer: Seagate
[99349.691102] usb 1-2: SerialNumber: NA9KR8KX
[99349.696871] usb 1-2: UAS is blacklisted for this device, using usb-storage instead
[99349.696876] usb-storage 1-2:1.0: USB Mass Storage device detected
[99349.697233] usb-storage 1-2:1.0: Quirks match for vid 0bc2 pid ab24: 800000
[99349.697405] scsi host1: usb-storage 1-2:1.0
[99350.760978] scsi 1:0:0:0: Direct-Access     Seagate  BUP Slim BK      0304 PQ: 0 ANSI: 6
[99350.762532] sd 1:0:0:0: Attached scsi generic sg0 type 0
[99350.763130] sd 1:0:0:0: [sda] Spinning up disk...
[99351.783604] ...ready
[99353.832593] sd 1:0:0:0: [sda] 3907029167 512-byte logical blocks: (2.00 TB/1.82 TiB)
[99353.864684] sd 1:0:0:0: [sda] Write Protect is off
[99353.864691] sd 1:0:0:0: [sda] Mode Sense: 4f 00 00 00
[99353.865555] sd 1:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[99353.969231]  sda: sda1
[99353.970796] sd 1:0:0:0: [sda] Attached SCSI disk
[99356.983169] EXT4-fs (dm-1): mounted filesystem with ordered data mode. Opts: (null)
```
But I cannot get the same to happen on the GnuBee: without `uas.ko`, no driver claims the device.

I guess time to file upstream at kernel.org, as suggested by @neheb above

neheb, 2019-02-14:
Something's not right. On my setup, I have usb-common.ko, usb-storage.ko and usbcommon.ko. I do not have uas.ko. USB works fine.

The issue that you're seeing is a compatibility issue between your USB device and mtk-usb (the USB driver).

cbm755, 2019-02-14:
Other non-UAS BOT-only drives do work fine.  Have you tested with a UAS-compatible drive?

neheb, 2019-02-14:
Yes. Works fine.

```
[94777.612500] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
[94777.654281] usb-storage 2-1:1.0: USB Mass Storage device detected
[94777.667267] scsi host6: usb-storage 2-1:1.0
[94778.713144] scsi 6:0:0:0: Direct-Access     Seagate  Expansion        0502 PQ: 0 ANSI: 6
[94778.732128] sd 6:0:0:0: [sdg] Spinning up disk...
[94779.752449] .
[94780.792454] .
[94780.795867] ready
[94780.803205] sd 6:0:0:0: [sdg] 1953525167 512-byte logical blocks: (1.00 TB/932 GiB)
[94780.926794] sd 6:0:0:0: [sdg] Write Protect is off
[94780.936386] sd 6:0:0:0: [sdg] Mode Sense: 4f 00 00 00
[94780.946939] sd 6:0:0:0: [sdg] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[94781.002535] sd 6:0:0:0: [sdg] Attached SCSI disk
```

modinfo shows
```
usb_storage            39867  0
```

on my laptop, uas is shown at the end.

neilbrown, 2019-02-15:
You could try to unbind the device from uas and bind to usb-storage.
echo $foo > /sys/bus/usb/drivers/uas/unbind
echo $foo > /sys/bus/usb/drivers/usb-storage/bind

$foo is some name listed in "ls /sys/bus/usb/devices". Maybe 1-4:1.0

cbm755, 2019-02-15:
Boot with `uas.ko` and drive attached:
```
[   14.139386] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   14.639039] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[   14.653920] ehci-platform: EHCI generic platform driver
[   14.678164] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[   14.691950] ohci-platform: OHCI generic platform driver
[   14.713059] uhci_hcd: USB Universal Host Controller Interface driver
[   14.746742] usbcore: registered new interface driver usb-storage
[   14.766031] scsi host6: uas
[   14.772540] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 6
[   14.789722] xhci-mtk 1e1c0000.xhci: @000000000f67e360 0ea78000 00000000 05000000 01078001
[   14.806354] xhci-mtk 1e1c0000.xhci: ERROR Transfer event for unknown stream ring slot 1 ep 2
[   14.823515] xhci-mtk 1e1c0000.xhci: @000000000f67e370 0ea78100 00000000 05000000 01038001
[   36.212082] scsi 6:0:0:0: tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN 
[   36.227518] scsi 6:0:0:0: tag#0 CDB: opcode=0x12 12 00 00 00 24 00
[   36.240201] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[   36.260195] xhci-mtk 1e1c0000.xhci: ep deq seg = 8ea73100, deq ptr = aea72010
[   37.377946] scsi host6: uas_eh_device_reset_handler FAILED to get lock err -16
[   37.392666] scsi 6:0:0:0: Device offlined - not ready after error recovery
[   37.407147] usbcore: registered new interface driver uas
[   37.418359] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
```

### uas unbind

```
# echo "2-1:1.0" > /sys/bus/usb/drivers/uas/unbind
# dmesg
  ---- 8< ----
[  256.773641] xhci-mtk 1e1c0000.xhci: Mismatch between completed Set TR Deq Ptr command & xHCI internal state.
[  256.793676] xhci-mtk 1e1c0000.xhci: ep deq seg = 8ea79880, deq ptr = af038010
```

### usb-storage bind:

```
# echo "2-1:1.0" > /sys/bus/usb/drivers/usb-storage/bind 
ash: write error: No such device
# dmesg
  << nothing new >>
```

neilbrown, 2019-02-15:
Looks like a bug somewhere - some error path not cleaning up I suspect.
What happens if you blacklist (or remove) the uas module?  Can you provide relevant dmesg lines for that scenario?

cbm755, 2019-02-15:
Remove `uas.ko` present, reboot, then hot plug the drive: 
```
# dmesg
 ---- 8< ----
[ 7540.277778] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
# echo "2-1:1.0" > /sys/bus/usb/drivers/usb-storage/bind 
ash: write error: No such device
# dmesg
 << nothing new >>
```
I reiterate that other non-uas (BOT only) drives work fine.

cbm755, 2019-02-15:
Maybe the `mtk-usb` only supports whichever interface is listed *first*?

@neheb said:
> on my laptop, uas is shown at the end.

On Fedora w/ 4.20.6, note `uas` comes first in the `lsusb -t` list.   So maybe the difference between our devices is the order in which they list their two `bInterfaceProtocol`.

```
# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
    |__ Port 3: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
```
@neheb could you post your `lsusb -t` output both on your laptop and your GnuBee?

My theory here is that on `mtk-usb`, it can only find `Port 1: Dev 3` (in my output above) and *not* `Port 3: Dev 2`.

neheb, 2019-02-16:
GnuBee
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
    |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
```
Laptop
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
```

Unless you have kmod-usb-storage-uas, you should not have uas at all. Maybe reinstall firmware?

aocana, 2019-04-06:
any update related with this issue?

cbm755, 2019-04-06:
Can you verify that it still doesn't work even if you remove the `uas.ko` module?

Otherwise, no updates from me.  Something should be reported upstream (at kernel.org as per ["Problems with mtk-usb should probably be directed at Mediatek over on kernel.org"](https://github.com/gnubee-git/GnuBee_Docs/issues/98#issuecomment-463343056).  But I haven't done that.
@neilbrown said "Looks like a bug somewhere - some error path not cleaning up I suspect."

Note there is a mistake in my comments about about Fedora: the device only appears once in `lsusb -t`: I was mistaken about this.

nyov, 2019-08-14:
drive-by comment (not involved with the project, don't know uas or have such a drive, only found this randomly)

It seems you may have ignored an important message from your Fedora:
```
usb 1-2: UAS is blacklisted for this device, using usb-storage instead
```
This indicates that Fedora has a workaround for this specific device. Searching the internet, for where this message might originate, you can find some interesting stuff.

This looked relevant:
https://github.com/hardkernel/linux/pull/291/files
which points to here (didn't read):
https://forum.odroid.com/viewtopic.php?f=146&t=26016&start=50#p188246
and looks like it's a kernel-level blacklisting:
https://github.com/torvalds/linux/blob/master/drivers/usb/storage/unusual_uas.h#L41-L46
(lists your device id)

As such, it seems you either need to get a newer kernel, the OpenWRT people to include the fix, or patch it yourself. HTH

cbm755, 2019-08-14:
Thanks.  I may have blacklisted it myself for that output, can't recall.  Will try to double-check next time I have this drive out.

cbm755, 2019-08-18:
Confirmed, the drive is **not** blacklisted out-of-the-box on Fedora x86:
```
[752088.921065] usb 2-2: new SuperSpeed Gen 1 USB device number 4 using xhci_hcd
[752088.934266] usb 2-2: New USB device found, idVendor=0bc2, idProduct=ab24, bcdDevice= 1.00
[752088.934271] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[752088.934275] usb 2-2: Product: BUP Slim BK
[752088.934278] usb 2-2: Manufacturer: Seagate
[752088.934280] usb 2-2: SerialNumber: NA9KR8KX
[752088.940890] scsi host1: uas
[752088.941976] scsi 1:0:0:0: Direct-Access     Seagate  BUP Slim BK      0304 PQ: 0 ANSI: 6
[752088.942902] sd 1:0:0:0: [sdb] Spinning up disk...
[752088.942918] sd 1:0:0:0: Attached scsi generic sg1 type 0
```
The output I had posted [above](https://github.com/gnubee-git/GnuBee_Docs/issues/98#issuecomment-463737654) was when I manually blacklisted it.

thanks @nyov for the interest, and the helpful links

andrey-kiselev, 2019-08-31:
> I manually blacklisted it

Could you please post instructions how to blacklist it? Thanks!

paulfertser, 2020-01-05:
> > I manually blacklisted it
> 
> Could you please post instructions how to blacklist it? Thanks!

Andrey, please try editing ''/etc/modules.d/usb-storage'' to add ''quirks=VID:PID:u'' after ''usb-storage'' (on the same line). VID/PID must be of your UAS device. ''kmod-usb-storage-uas'' should be installed. Reboot for the changes to take effect.