gnubee-git/GnuBee_Docs #100: Poor cryptsetup performance

dr4Ke, 2019-06-26:
I'm trying to set up a luks encrypted volume on GB-PC2. The performances seem low.
```
# cryptsetup benchmark
# Tests are approximate using memory only (no storage IO).
PBKDF2-sha1        63503 iterations per second for 256-bit key
PBKDF2-sha256      74727 iterations per second for 256-bit key
PBKDF2-sha512      18778 iterations per second for 256-bit key
PBKDF2-ripemd160   56790 iterations per second for 256-bit key
PBKDF2-whirlpool   12365 iterations per second for 256-bit key
#  Algorithm | Key |  Encryption |  Decryption
     aes-cbc   128b    11.0 MiB/s    11.0 MiB/s
 serpent-cbc   128b     9.3 MiB/s     9.5 MiB/s
 twofish-cbc   128b    11.7 MiB/s    11.4 MiB/s
     aes-cbc   256b     8.6 MiB/s     8.5 MiB/s
 serpent-cbc   256b     9.3 MiB/s     9.5 MiB/s
 twofish-cbc   256b    11.7 MiB/s    11.4 MiB/s
     aes-xts   256b    11.1 MiB/s    10.9 MiB/s
 serpent-xts   256b     9.6 MiB/s     9.6 MiB/s
 twofish-xts   256b    12.0 MiB/s    11.5 MiB/s
     aes-xts   512b     8.7 MiB/s     8.5 MiB/s
 serpent-xts   512b     9.6 MiB/s     9.6 MiB/s
 twofish-xts   512b    12.0 MiB/s    11.5 MiB/s
```
Is this expected?
Is this realistic to use disk encryption on this device?

Sorry if that's not the good place to ask.

neheb, 2019-06-26:
There's an alpha level driver available that boosts the speed up to ~40MB/s. Still slow.

So no.

Qwertie-, 2019-08-05:
Would disk encryption be useful on a nas? You would have to leave the key unencrypted on the device for it to recover from a power outage. Might be more useful to use client programs that encrypt data before sending it to the device.

kripton, 2020-01-09:
> Would disk encryption be useful on a nas? You would have to leave the key unencrypted on the device for it to recover from a power outage.

Sure it's useful. You don't have to have the key on the device itself. Just have a service that asks *other* machines on the network for the key and "decrypt"/mount the volumes then.
Doing the data en/decryption on the client machines could make the "access files from multiple machines simultaniously" more complicated, depending on which encryption technology you use (block-level vs. file based)

cbm755, 2020-01-09:
For me, its also the idea that I can just recycle an old harddrive without smashing it with a hammer first :)