gnubee-git/GnuBee_Docs #103: Local ip address changes on every reboot.

Qwertie-, 2019-08-05:
This seems like a bit of a strange issue and I'm not sure the gnubee is at fault but no other devices on my network seem to do this. Every time the gnu bee reboots it gets a new local IP address. My routers device page shows a new listing for the gnu bee added every time it boots. Has anyone else had this issue?

xvybihal, 2019-08-05:
Just a wild guess. Do you have same MAC or it also always change? 
I think I had this long time ago. What kernel you using?

Qwertie-, 2019-08-05:
Kernel is `Linux gnubee.gnubee 4.4.172 #5 SMP Mon Feb 4 07:45:13 UTC 2019 mips GNU/Linux` using the Neil brown build. I'll have to keep an eye on the mac address. Thats probably what's tripping up my router.

Qwertie-, 2019-08-07:
Yeah I can see the mac address changes on every reboot

neheb, 2019-08-07:
probably need new firmware. mac address can be edited in the bootloader menu.

Qwertie-, 2019-08-07:
@neheb Is there any way to update it or do I have to reinstall?

Qwertie-, 2019-08-10:
This was fixed after installing the 5.2 firmware

closed 2019-08-10 by Qwertie-