gnubee-git/GnuBee_Docs #107: Does the GnuBee 1 have 1Gbps per port, or overall?

kozross, 2019-12-29:
I was investigating using link aggregation on the GnuBee 1. However, it appears to me that the GnuBee has a single NIC connected to a two-port switch. By that, I'm guessing that I can only get 1Gbps overall, _not_ per port. Am I correct?

pinney, 2019-12-29:
Correct on the PC1 they are connected to the switch.

kozross, 2019-12-29:
So does this mean I'm correct? I was a bit confused, because the [GnuBee PC1 specifications page on CrowdSupply][1] describes it as having '2 x Gigabit Ethernet', which I assumed meant I could get 1Gbps on each port. My plan was to link aggregate them, but I guess there's not much point if they're shared like that.

pinney, 2019-12-29:
It's the same as most MT7621 routers.
They are connected to the MT7530 switch.

vincele, 2022-12-06:
I tested with 2 x iperf3 clients, each bound to one eth interface of gnubee PC1, through a 1gbps netgear switch, to 2 other separate hosts. No process CPU binding, though.

The switch can handle 1gbps per port easily (tested with 2 parallel streams between 2 hosts pairs 927Mbps + 896Mbps).

I first tested iperf3 -c to each host sequentially, to ensure both pipes can achieve the 1gbps.
Both got pretty identical 937 Mbits/sec from the GBPC1 clients.

Doing then both in parallel, they shared the BW: 467Mbps & 471Mbps. Which is IMHO a HW limitation of the platform.

Case closed ?
:-)