gnubee-git/GnuBee_Docs #108: Are PCB schematics open source too?

bornlibra23, 2020-04-30:
I would like to print the board myself & try to upgrade the number of sata ports. This may involve changing the components on the main board & the number of PCI-E lanes.
Is the schematic available for me to work off?

bornlibra23, 2020-04-30:
I found this. Isnt there some form of a hardware definition file which can be directly printed into a PCB like an input file for a 3D printer?
https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PCx/Documents/GB-PC2_V1.1_schematic.pdf

jjakob, 2020-05-01:
If there is a board layout or schematic in raw form, I haven't been able to find it. Highly likely just the company that did the design and manufacturing has it and only released the schematic in PDF form.