gnubee-git/GnuBee_Docs #110: GB-PC2 needs IP1001 in its devicetree to make use of the yellow eth phy

k3ninho, 2020-10-21:
I was hoping to bond two GBe devices to drive the array of disks on my GB-PC2 from another machine using iSCSI, iSER or SRP. However, the black and blue ports share a 1GBps uplink from the switch portion to the CPU portion within the mt7621 but there's second gigabit rgmii link to the IP1001 Ethernet transceiver chip for the yellow physical port.

(Huge props to Neil Brown for the build guide at https://github.com/neilbrown/gnubee-tools and the ongoing refinements to his firmware for gb-pc1 and gb-pc2. I want to find the interrupt storm or context switching nightmare that holds back the mt7621 CPU from performing at its best.)

Page 17 of the schematic for the GB-PC2 shows the third ethernet phy connected to the IP1001 chip you can see when you turn over the board. There's a rgmii defined for the GB-PC1 and GB-PC2 on page 13 of the schematic with GE2 outputs going to GE family shown on the IP1001 schematic of  p17. 

The mt7621.dtsi file says that gmac1 is that uplink and there's signaling pins in the mdio-bus for &rgmii2_pins associated with gmac1. That's why I think that the IP1001 is on gmac1 and there's even a driver in the kernel tree at /drivers/net/phy/icplus.c that references initialising the ip1001. 

The gb-pc2.dtsi needs to have the right plumbing for this Ethernet transceiver so that users of GnuBee PC-2 can have a pair of independent GBe ports in use -- I think this is what needs to be done but I'm unsure how to go about it.

neilbrown, 2021-01-01:
It is my understanding that the second network interface in the MT7621 already connects through the ip1001 to the yellow port on the GB-PC2, and on the GB-PC1 it is not connected to anything.
Certainly with my gbpc2 firmware I can use "eth1" to communicate on the yellow port, and I'm quite certain that isn't going through the internal switch.
Or did I misunderstand your question?

k3ninho, 2021-01-04:
Hey Neil! Happy New Year to all!

I've not managed to get the yellow ports on either of my two GBPC2s to work using your firmware. Neither a firmware downloaded from your site or home-built cross-compile let me use the yellow phy. (BTW it was easy to cross-compile following your instructions -- the instructions are good, thank you!)

The goal is to have one GBe phy using a full gigabit link to the mt7621 bonded to a second GBe phy on a second full gigabit link to the MT7621 chip, hoping to use the GBPC2 as an iSCSI disk target for targetcli. That's why I attempted the homebuild to enable some of the in-kernel config options so I can take advantage of targetcli, if any, over userspace tgt controls.

So I looked at the devicetree as per the git trees I've cloned and are using to build, and think that it lacks an explicit ip1001 block connected to the gmac1, expecting that the ip1001 needs to be woken up to let it pass data through. 

Using your GBPC2 firmware means yellow port should be working on my boards -- but 'should' is tricksy and I don't get any notifications that there's a connector in the eth1 port or that a link might be available/has come up. I note other commentary online says that the LED indicators aren't to be trusted so I'm not concerned that the yellow port LED's don't light up -- but it might be helpful infor for troubleshooting here. 

Do you have any extra steps I might take to work out what's obstructing this port?

Cheers,
K3n.